USBGuard/usbguard #578: configure: error: Link test failed both with and without -lstdc++fs; something is broken

commandline-be, 2022-12-19:
Compilation fails on RHEL9 derivative (Rocky Linux 9.1)

git clone -b usbguard-1.1.2 https://github.com/USBGuard/usbguard.git
cd usbguard
dnf install dh-autoreconf libqb-devel audit-libs-devel libseccomp-devel libcap-ng-devel protobuf-devel  
sh autogen.sh
./configure --enable-seccomp --enable-asan --enable-audit --enable-libcapng --enable-static --enable-tsan --enable-systemd --with-dbus --with-polkit --with-ldap  --with-crypto-library=openssl --disable-umockdev --with-bundled-catch

....
configure: Using bundled Catch library
checking whether we need to link to -lstdc++fs for PEGTL explicitly... ERROR
configure: error: Link test failed both with and without -lstdc++fs; something is broken, please check file config.log for details.

-----

unclear why, assuming it could be because libstdc++ = 11.x

similar thread here https://lore.kernel.org/all/20220820124154.GJ2167049@scaer/T/

hartwork, 2022-12-19:
Hi @commandline-be which compiler and compiler version are you using? If this GCC prior than version 8 then it would probably be unsupported. If it's either Clang or GCC >=8 please share the full `./configure` output and file `config.log`. Thank you!

commandline-be, 2022-12-19:
hey, here it is

gcc --version
gcc (GCC) 11.3.1 20220421 (Red Hat 11.3.1-2)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

hartwork, 2022-12-19:
@commandline-be could you share the full `./configure` output and resulting file `config.log`? Thank you!

commandline-be, 2022-12-19:
Sorry, misread that line. Here i share the output with the ./configure invokation, i tried multiple ./configure lines including just running './configure --with-bundled-catch --enable-full-test-suite' only. The fail is at the same point for each attempt.

---- from the build directory or from above the fail situation is unchanged ---

../configure --enable-seccomp --enable-asan --enable-audit --enable-libcapng --enable-static --disable-shared --enable-tsan --enable-systemd --with-dbus --with-polkit --with-ldap  --with-crypto-library=openssl --disable-umockdev --with-bundled-catch --without-bundled-pegtl --enable-full-test-suite 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking whether C compiler accepts -Wno-implicit-fallthrough... yes
checking for g++... no
checking for c++... no
checking for gpp... no
checking for aCC... no
checking for CC... no
checking for cxx... no
checking for cc++... no
checking for cl.exe... no
checking for FCC... no
checking for KCC... no
checking for RCC... no
checking for xlC_r... no
checking for xlC... no
checking whether we are using the GNU C++ compiler... no
checking whether g++ accepts -g... no
checking dependency style of g++... none
checking for gcc option to accept ISO C99... none needed
checking whether make sets $(MAKE)... (cached) yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
[... 1283 lines cut ...]
#define STDC_HEADERS 1
#define HAVE_SYS_TYPES_H 1
#define HAVE_SYS_STAT_H 1
#define HAVE_STDLIB_H 1
#define HAVE_STRING_H 1
#define HAVE_MEMORY_H 1
#define HAVE_STRINGS_H 1
#define HAVE_INTTYPES_H 1
#define HAVE_STDINT_H 1
#define HAVE_UNISTD_H 1
#define HAVE_DLFCN_H 1
#define LT_OBJDIR ".libs/"
#define HAVE_PTHREAD_PRIO_INHERIT 1
#define HAVE_PTHREAD 1
#define HAVE_UEVENT 1
#define HAVE_LIBQB 1
#define HAVE_LIBSODIUM 1
#define HAVE_LIBCRYPTO 1
#define USBGUARD_USE_OPENSSL 1
#define HAVE_LINUX_AUDIT 1
#define HAVE_DECL_AUDIT_ENCODE_NV_STRING 1
#define HAVE_SECCOMP 1
#define HAVE_LIBCAPNG 1

configure: exit 1

hartwork, 2022-12-19:
@commandline-be multiple occurrences of `g++: command not found` caught my eyes. I believe that means two things:
- That you have a C compiler installed — `${CC}`/`gcc` — but not a C++ compiler — `${CXX}`/`g++`
- That the configure script should bail out earlier and be more helpful about it.

What do you think?

commandline-be, 2022-12-19:
@hartwork Thanks, true that, `dnf install g++ ` fixed the 'Link test failed both with and without' issue.

I agree to make reporting on missing commands more helpful, i overlooked that 3 times or so.

commandline-be, 2022-12-19:
FYI: for RHEL systems `dnf install dh-autoreconf libqb-devel audit-libs-devel libseccomp-devel libcap-ng-devel protobuf-devel polkit-devel asciidoc openldap-devel` install most if not all dependencies.

Notable is umockdev does not seem to have a suitable package on RHEL.

commandline-be, 2022-12-19:
Thanks.  My reason for mentioning the dnf line is it is not included in the Readme :-) Including it would save some time.

I don't feel confident to address the proposed work around for --enable-asan. I was looking into compilation following a response/request on the Rocky Linux Forums.

I don't fully understand what asan and tsan offer and what makes one preferable over the other or their efficacy on an SELinux enabled system.

hartwork, 2022-12-19:
> Thanks. My reason for mentioning the dnf line is it is not included in the Readme :-) Including it would save some time.

We could add it, but then we cannot do that for every distro on the planet, as it does not scale well and needs to e kept in sync :man_shrugging:

> I don't feel confident to address the proposed work around for --enable-asan. I was looking into compilation following a response/request on the Rocky Linux Forums.
>
> I don't fully understand what [...]

We're going a bit off-topic here if I join in, so I'll just say that ASan detects things like out-of-bounds array access at runtime. It's one of the best tools that C/C++ developers have, in my opinion. ASan rocks, big time.

commandline-be, 2022-12-19:
Thanks, works for me.

 I'll look into asan and how to get it to compile for USBGuard on RHEL9.

closed 2022-12-19 by commandline-be