USBGuard/usbguard #579: USB-C behavior & potential bypasses

pdolinic, 2022-12-23:
Even though USBC connected devices show as blocked, they appear visible on the device tree, (try it via a phone USB-C -> USB-C)

Some investigation might be worth there, and if one could exploit simply adding USBC Adapters on top of USB-A/USB-B.

The first time I found this interesting, was when playing around with USB-C Yubikeys.

hartwork, 2022-12-23:
Hi @pdolinic, maybe its just me but I think a few pictures and/or shell output would help to better understand the blocked-and-not-blocked situation that you're describing. Could you demonstrate for a particular device what parts of the system show that's blocked in detail and what parts don't? E.g. I'm unsure if we're talking a mounted file system here or more existence of a `/dev/foo` block device and so on. More details would be great. Thank you!

pdolinic, 2022-12-24:
Hey @hartwork , thanks for replying

So I've tested  
a) a mixed USB-C Headphone from Logitech & everything seems to be as expected (seeing it block in the terminal, and looking for it the sound output it is not appearing as well)
b)  but on the test-phone, I've connected it via USB-C I can see it in the file manager appear, even before having given it any unblock via ` usbguard-add device xyz` <-> I assume this is just cosmetic? 

[image]

State:
```
17: block id 18d1:4ee1 serial "strippedXYZ" name "Pixel 6 Pro" hash "strippedXYZ" parent-hash "strippedXYZ" via-port "3-4" with-interface 06:01:01 with-connect-type "hotplug"
```
- Maybe this is just cosmetic, are there any paths /ways  I could check? 
- Even when going on the phone on "USB Preferences -> File transfer" I cannot access the data, so USBGuard appears to be working
- Still the question remains, why it is showing up on devices?

Thanks

commandline-be, 2023-11-25:
could it be you do see the device show up but you cannot browse the filesystem(s) ?
i think that's what I've seen before and should be 'normal behavior'