USBGuard/usbguard #581: [question] What is a "parent-hash"

TheFloatingBrain, 2023-01-19:
Hello, 

    I had some devices which I set rules for with the `-p` flag. However, they stopped looking, I compared the devices stats to what was in the rules, and there hashes were the same, but their parent hashes were now different, I was wondering what a parent-hash was and if I can assume the device was modified?

Thank you, 
 - TFB

muelli, 2023-01-19:
I think it refers to the port the device is attached.

The Changelog says:
```
 - Implemented "parent-hash" attribute for associating a device with its
  parent device.- Implemented "parent-hash" attribute for associating a device with its
  parent device.
```

TheFloatingBrain, 2023-01-20:
> 

Thanks!

Huh' its weird, I am using an adapter, but the two adapters I have have the same parent - hash and they are both different than the parent hash listed in the rules... I wonder what changed.