USBGuard/usbguard #585: USBGuard failing to block new devices

johncarterofmars, 2023-03-26:
Hello. I am using USBguard on Mint 20.3. I have used it for years without issue but in the last week, something weird started to happen. I decided to purge and reinstall. I generated a new policy with all the devices plugged in and it all seemed fine. However, when I plugged in a flash drive that wasn't on the approved list, Linux mounted it anyway. I moved it to different ports on the system and hub and it mounted on all. 

My implicit policy is set to block: ImplicitPolicyTarget=block

Here is the relevant log file:

`[1679846097.932] (A) uid=0 pid=11345 result='SUCCESS' device.rule='block id 13fe:6400 serial "070D13CF3668E724" name "USB DISK 3.0" hash "uXskSTsiyHnhR08zEoWblI126JoSKiyS7RJ+SiGdKkA=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.3.2" with-interface 08:06:50 with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2/2-4.2.1/2-4.2.1.3/2-4.2.1.3.2' type='Device.Insert' 
[1679846097.932] (A) uid=0 pid=11345 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2/2-4.2.1/2-4.2.1.3/2-4.2.1.3.2' target.new='block' device.rule='block id 13fe:6400 serial "070D13CF3668E724" name "USB DISK 3.0" hash "uXskSTsiyHnhR08zEoWblI126JoSKiyS7RJ+SiGdKkA=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.3.2" with-interface 08:06:50 with-connect-type "unknown"' target.old='block' type='Policy.Device.Update' 
[1679846102.146] (A) uid=0 pid=11345 result='SUCCESS' device.rule='block id 13fe:6400 serial "070D13CF3668E724" name "USB DISK 3.0" hash "uXskSTsiyHnhR08zEoWblI126JoSKiyS7RJ+SiGdKkA=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.3.2" with-interface 08:06:50 with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2/2-4.2.1/2-4.2.1.3/2-4.2.1.3.2' type='Device.Remove' `

I also verified that this device is not in rules.conf. In fact, there aren't any USB DISKS listed. 

I am sure this is something I have done incorrectly. Would anyone have any suggestions of where to begin looking? 
Thank you.

hartwork, 2023-03-26:
Hi @johncarterofmars,

https://community.linuxmint.com/software/view/usbguard doesn't seem to list a version. Could you share the output of `apt-cache policy usbguard` so we''ll know which version you have?

PS: Could you insert a newline before `[1679846097.932]` and `[1679846102.146]` to make it easier to see that it's three lines?

Thanks and best, Sebastian

johncarterofmars, 2023-03-27:
sure, sorry about that. I first installed 0.7.6 from the Ubuntu repo. When that didn't work, I tried to build it but was unsuccessful. So I hunted down a 1.0 deb file and got that installed. Service starts, runs, I can run all usbguard commands, etc. Here is the output:

`$ apt-cache policy usbguard
usbguard:
  Installed: 1.0.0+ds-2
  Candidate: 1.0.0+ds-2
  Version table:
 *** 1.0.0+ds-2 100
        100 /var/lib/dpkg/status
     0.7.6+ds-1build1 500
`

muelli, 2023-03-27:
This is prone to fail.
You need libusbguard, too.
Probably other dependencies.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

johncarterofmars, 2023-03-27:
I installed each dependency as well. libprorobuf, libusbguard, libqt, etc. But even if I missed something, it doesn't explain why the initial install of 0.7.6 wasn't working, right?

hartwork, 2023-03-27:
@johncarterofmars I can offer to jump on a voice call with screen sharing and we do the same thing again with latest Git master: we'd start out with an empty rules file, auto-add all things connect, get the flash drive in, and see if we can figure things out. If that would help in some way, please drop me a mail though my profile e-mail. If it's too much, no problem either.

johncarterofmars, 2023-03-28:
If it leads to that, then so be it, but I would prefer not to have to do that. Is there anything else you'd like me to try first?

hartwork, 2023-03-28:
@johncarterofmars currently I have no idea what may be going on so it it was just an idea in hope it could help clear the fog. We can wait for other ideas, let's see what others think.

muelli, 2023-03-28:
dmesg could be instructive, too.

johncarterofmars, 2023-03-29:
No worries. 

`dmesg | grep usb
[243243.506272] usb 2-4.2.1.3.2: new SuperSpeed USB device number 22 using xhci_hcd
[243243.528651] usb 2-4.2.1.3.2: New USB device found, idVendor=13fe, idProduct=6400, bcdDevice= 1.00
[243243.528664] usb 2-4.2.1.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[243243.528670] usb 2-4.2.1.3.2: Product: USB DISK 3.0
[243243.528673] usb 2-4.2.1.3.2: Manufacturer:         
[243243.528677] usb 2-4.2.1.3.2: SerialNumber: 070D13CF3668E724
[243243.528948] usb 2-4.2.1.3.2: Device is not authorized for usage
[243243.555720] usb-storage 2-4.2.1.3.2:1.0: USB Mass Storage device detected
[243243.556229] scsi host5: usb-storage 2-4.2.1.3.2:1.0
[243243.556376] usb 2-4.2.1.3.2: authorized to connect
[243274.314703] usb 2-4.2.1.3.2: USB disconnect, device number 22
`

johncarterofmars, 2023-04-01:
I decided to keep testing. I restarted the service and plugged the flash drive in, and then ran:
`usbguard list-devices | grep block
52: block id 13fe:6400 serial "070D13CF3668E724" name "USB DISK 3.0" hash "uXskSTsiyHnhR08zEoWblI126JoSKiyS7RJ+SiGdKkA=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.3.3" with-interface 08:06:50 with-connect-type "unknown"
`
Ok so it least it sees it here but the weird part is that Mint still saw and mounted the drive. Even though it was listed by USBguard and being blocked. 
Here are the USBGuard logs of the event:

[1680306232.389] (A) uid=0 pid=1782022 result='SUCCESS' device.rule='block id 0781:55a9 serial "010108766b5b5164130af1aaa4af81959f7ce4957036ef08ed99619447989f4d65e000000000000000000000a053fb9aff8f4100a95581077aaa395d" name " SanDisk 3.2Gen1" hash "kEf6sEgsKojlfZmJ/99Us1r96PsJyk5BqBo8wjgZtKo=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.1" with-interface 08:06:50 with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2/2-4.2.1/2-4.2.1.1' type='Device.Insert' 

[1680306232.389] (A) uid=0 pid=1782022 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2/2-4.2.1/2-4.2.1.1' target.new='block' device.rule='block id 0781:55a9 serial "010108766b5b5164130af1aaa4af81959f7ce4957036ef08ed99619447989f4d65e000000000000000000000a053fb9aff8f4100a95581077aaa395d" name " SanDisk 3.2Gen1" hash "kEf6sEgsKojlfZmJ/99Us1r96PsJyk5BqBo8wjgZtKo=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.1" with-interface 08:06:50 with-connect-type "unknown"' target.old='block' type='Policy.Device.Update' 

[1680306260.865] (A) uid=0 pid=1782022 result='SUCCESS' device.rule='block id 0781:55a9 serial "010108766b5b5164130af1aaa4af81959f7ce4957036ef08ed99619447989f4d65e000000000000000000000a053fb9aff8f4100a95581077aaa395d" name " SanDisk 3.2Gen1" hash "kEf6sEgsKojlfZmJ/99Us1r96PsJyk5BqBo8wjgZtKo=" parent-hash "EYAl1sYHGTzLpQKDkXPT5hZoxsDGkWVQLP02mJdPwjI=" via-port "2-4.2.1.1" with-interface 08:06:50 with-connect-type "unknown"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2/2-4.2.1/2-4.2.1.1' type='Device.Remove' 

So from what I can tell, USBGuard thinks its working but Mint disagrees and mounts the drive anyway.