USBGuard/usbguard #588: Failed to permanently add device to rules.conf: Rule set upsert: rule: Cannot upsert; multiple matching rules

genodeftest, 2023-04-15:
When I try to add a specific device to the allowlist, I get an error:
```
IPC ERROR: request id=1: Rule set upsert: rule: Cannot upsert; multiple matching rules
```

# Steps I followed:
1. get list of devices ([censored output of  `usbguard list-devices`](https://github.com/USBGuard/usbguard/files/11238918/usbguard_list-devices.censored.txt))
2. Allow device 16: `usbguard allow-device --permanent 16`

# What happens:
```
$ usbguard allow-device --permanent 16
IPC ERROR: request id=1: Rule set upsert: rule: Cannot upsert; multiple matching rules
```

# What should happen:
Work fine without errors.

# Additional info:
* Allowing the device without the `--permanent` flag works fine (just for the session)
* My USB device list (see attachment above) looks weird. Devices 15 and 16 have the same ID, an empty serial number, empty name and two **different**(!) parents.
* After enabling device 16 (non-permanent) and double-checking that it is listed as _allow_ in `usbguard list-devices`, `lsusb` shows the device only once ([output of `lsusb`](https://github.com/USBGuard/usbguard/files/11238929/lsusb.txt)). Also, `lsusb` shows the same output before and after allowing device 16, i.e., the kernel seems to see it as only one device. As a non-expert, this looks to me as if 15 and 16 are physically the same USB device, just with two different parents.
* I have been using the same physical hardware (no separate USB hubs attached) for a few years with the same linux distribution (Fedora, not reinstalled for years) and same USBGuard configuration without problems. According to my package manager's history (`dnf history list usbguard`), USBGuard has not been updated since 2022-09, but I've only started seeing this issue since a few weeks now. The hardware is very old and thus probably hasn't seen a firmware update (also, I couldn't find any in my syslog), so the issue is probably related to a recent software update, most probably the kernel.

# Software versions:
* usbguard-1.1.0-4.fc37.x86_64
* kernel-6.2.9-200.fc37.x86_64

genodeftest, 2023-04-15:
Additional notes regarding kernel versions:
1. With Fedora's kernel 6.2.7, everything is fine and the device shows up only once.
2. With Fedora's kernel 6.2.8, the device started showing up twice in `usbguard list-devices` with two different parent hashes
  * Side note: The enumeration order changed (but not the hashes), which in turn made **all** USB devices blocked with my configuration, but that's something I should probably report separately…
4. With Fedora's kernel 6.2.9, the device still shows up twice in `usbguard list-devices`.
  * Side note: With the enumeration order changed again, probably back to the order of 6.2.7 as far as I can tell.

pouek, 2023-09-08:
I had the same bug before and until today on debian 12, kernel 6.1.0-11-amd64, and usbguard 1.1.2+ds-3+b1.
I just found to fix this by reloading usbguard service with ;
``` sudo systemctl restart usbguard ```
I hope this helps.