USBGuard/usbguard #593: usbguard won't start

lucamaf, 2023-08-02:
I have installed usbguard 1.0.0 on rhel 9.2, follow the standard procedure to generate default rules and tried to start the service up but it fails miserably with the following error:

```
Aug 02 10:00:39 edge-nuc usbguard-daemon[2311803]: Rules: SourceLOCAL::/etc/usbguard/rules.conf: usbguard::Exception
Aug 02 10:00:39 edge-nuc systemd[1]: usbguard.service: Control process exited, code=exited, status=1/FAILURE
Aug 02 10:00:39 edge-nuc systemd[1]: usbguard.service: Failed with result 'exit-code'.
Aug 02 10:00:39 edge-nuc systemd[1]: Failed to start USBGuard daemon.
```

the weird thing is that the daemon itself with the same rules starts fine I think (the same goes if I try to execute the start daemon process command of the service: /usr/sbin/usbguard-daemon -f -s -K -c /etc/usbguard/usbguard-daemon.conf ): 
```
[luca@edge-nuc ~]$ sudo usbguard-daemon -c /etc/usbguard/usbguard-daemon.conf
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.rule='allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb1" with-interface 09:00:00 with-connect-type ""' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1' type='Device.Present'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1' target.new='allow' device.rule='allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb1" with-interface 09:00:00 with-connect-type ""' target.old='allow' type='Policy.Device.Update'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.rule='allow id 1d6b:0003 serial "0000:00:14.0" name "xHCI Host Controller" hash "prM+Jby/bFHCn2lNjQdAMbgc6tse3xVx+hZwjOPHSdQ=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb2" with-interface 09:00:00 with-connect-type ""' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2' type='Device.Present'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb2' target.new='allow' device.rule='allow id 1d6b:0003 serial "0000:00:14.0" name "xHCI Host Controller" hash "prM+Jby/bFHCn2lNjQdAMbgc6tse3xVx+hZwjOPHSdQ=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb2" with-interface 09:00:00 with-connect-type ""' target.old='allow' type='Policy.Device.Update'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.rule='allow id 8087:0029 serial "" name "" hash "ATK8pCmQtUYaUnwqUVuYssrOMkW8pdCSdZO4OC6zEtg=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-5" with-interface { e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 } with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1/1-5' type='Device.Present'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1/1-5' target.new='allow' device.rule='allow id 8087:0029 serial "" name "" hash "ATK8pCmQtUYaUnwqUVuYssrOMkW8pdCSdZO4OC6zEtg=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-5" with-interface { e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 } with-connect-type "hotplug"' target.old='allow' type='Policy.Device.Update'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.rule='allow id 1a40:0101 serial "" name "USB2.0 HUB" hash "an+hPjkSqC/s+AnuuG9Ke1ycUY5865rBmC3TWH/SXso=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-6" with-interface 09:00:00 with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1/1-6' type='Device.Present'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1/1-6' target.new='allow' device.rule='allow id 1a40:0101 serial "" name "USB2.0 HUB" hash "an+hPjkSqC/s+AnuuG9Ke1ycUY5865rBmC3TWH/SXso=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-6" with-interface 09:00:00 with-connect-type "hotplug"' target.old='allow' type='Policy.Device.Update'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.rule='allow id 0413:0308 serial "" name "" hash "GlU8N4oZzvftdlyL2HhCfvcsrB89jOD8zQlqbsFXj1M=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-10" with-interface { 03:01:01 03:01:02 } with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1/1-10' type='Device.Present'
[1690963436.759] (A) uid=0 pid=2313010 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb1/1-10' target.new='allow' device.rule='allow id 0413:0308 serial "" name "" hash "GlU8N4oZzvftdlyL2HhCfvcsrB89jOD8zQlqbsFXj1M=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-10" with-interface { 03:01:01 03:01:02 } with-connect-type "hotplug"' target.old='allow' type='Policy.Device.Update'
```

this is the rule file:
allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" with-interface 09:00:00 with-connect-type ""
allow id 1d6b:0003 serial "0000:00:14.0" name "xHCI Host Controller" hash "prM+Jby/bFHCn2lNjQdAMbgc6tse3xVx+hZwjOPHSdQ=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" with-interface 09:00:00 with-connect-type ""
allow id 8087:0029 serial "" name "" hash "ATK8pCmQtUYaUnwqUVuYssrOMkW8pdCSdZO4OC6zEtg=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-5" with-interface { e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 } with-connect-type "hotplug"
allow id 1a40:0101 serial "" name "USB2.0 HUB" hash "an+hPjkSqC/s+AnuuG9Ke1ycUY5865rBmC3TWH/SXso=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-6" with-interface 09:00:00 with-connect-type "hotplug"
allow id 0413:0308 serial "" name "" hash "GlU8N4oZzvftdlyL2HhCfvcsrB89jOD8zQlqbsFXj1M=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "1-10" with-interface { 03:01:01 03:01:02 } with-connect-type "hotplug"

Cropi, 2023-08-02:
Executing ```journalctl -xe -u usbguard``` should give you some hint on the root cause of the problem.
What permissions/ownership does the rules file have?

lucamaf, 2023-08-02:
permissions:
-rw-------. 1 luca luca

radosroka, 2023-08-02:
What about SELinux? What is the context of the file? Are there any AVCs?

lucamaf, 2023-08-02:
unconfined_u:object_r:user_home_t:s0 /etc/usbguard/rules.conf

Cropi, 2023-08-02:
I believe that the owner/group is not part of the IPCAllowedUsers/IPCAllowedGroups group. There are two ways on how to configure IPC access control, please refer to the IPC ACCESS CONTROL section inside the usbguard-daemon.conf man page.

radosroka, 2023-08-02:
> unconfined_u:object_r:user_home_t:s0 /etc/usbguard/rules.conf

It should be 
```
-rw-------.   1 root root system_u:object_r:usbguard_rules_t:s0     0 Jul 23  2022 rules.conf
```

lucamaf, 2023-08-02:
I changed it to 
-rw-------. 1 root root unconfined_u:object_r:user_home_t:s0 1186 Aug  2 10:06 /etc/usbguard/rules.conf

and still not being able to start the service, I already did daemon reload

lucamaf, 2023-08-02:
getting the following error:
`restorecon: SELinux: Could not get canonical path for allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" with-interface 09:00:00 with-connect-type "" restorecon:  No such file or directory.`

radosroka, 2023-08-02:
> getting the following error: `restorecon: SELinux: Could not get canonical path for allow id 1d6b:0002 serial "0000:00:14.0" name "xHCI Host Controller" hash [...]

Sorry, I made a mistake. -f does something different.

```
sudo restorecon -v /etc/usbguard/rules.conf 
```

lucamaf, 2023-08-02:
that did the trick!

closed 2023-08-02 by radosroka