USBGuard/usbguard #594: Device blocked after suspend despite being allowed in rules

meden, 2023-08-05:
Hello, I'm facing the following issue: when resuming the system from suspend, one of the internal devices (a Synaptics fingerprint reader, with USB id `06cb:00f9`) gets blocked, despite the rules allow it.

I'm using USBguard 1.1.2 in Debian 12.

Here the log, where I (1) restart the service, (2) suspend&resume, (3) restart the service again:

```
ago 05 11:08:12 alg-pc systemd[1]: Stopping usbguard.service - USBGuard daemon...
ago 05 11:08:12 alg-pc systemd[1]: usbguard.service: Deactivated successfully.
ago 05 11:08:12 alg-pc systemd[1]: Stopped usbguard.service - USBGuard daemon.
ago 05 11:08:12 alg-pc systemd[1]: Starting usbguard.service - USBGuard daemon...
ago 05 11:08:12 alg-pc systemd[1]: Started usbguard.service - USBGuard daemon.
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 1d6b:0002 serial "XXXXXX" name "xHCI Host Controller" hash "d3YN7OD60Ggqc9hClW0/al6tlFEshidDnQKzZRRk410=" parent-hash "Y1kBdG1uWQr5CjULQs7uh2F6pHgFb6VDHcWLk83v+tE=" via-port "usb1" with-interface 09:00:00 with-connect-type ""' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb1' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb1' target.new='allow' device.rule='allow id 1d6b:0002 serial "XXXXXX" name "xHCI Host Controller" hash "d3YN7OD60Ggqc9hClW0/al6tlFEshidDnQKzZRRk410=" parent-hash "Y1kBdG1uWQr5CjULQs7uh2F6pHgFb6VDHcWLk83v+tE=" via-port "usb1" with-interface 09:00:00 with-connect-type ""' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 1d6b:0003 serial "XXXXXX" name "xHCI Host Controller" hash "4Q3Ski/Lqi8RbTFr10zFlIpagY9AKVMszyzBQJVKE+c=" parent-hash "Y1kBdG1uWQr5CjULQs7uh2F6pHgFb6VDHcWLk83v+tE=" via-port "usb2" with-interface 09:00:00 with-connect-type ""' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb2' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:0d.0/usb2' target.new='allow' device.rule='allow id 1d6b:0003 serial "XXXXXX" name "xHCI Host Controller" hash "4Q3Ski/Lqi8RbTFr10zFlIpagY9AKVMszyzBQJVKE+c=" parent-hash "Y1kBdG1uWQr5CjULQs7uh2F6pHgFb6VDHcWLk83v+tE=" via-port "usb2" with-interface 09:00:00 with-connect-type ""' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 1d6b:0002 serial "XXXXXX" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb3" with-interface 09:00:00 with-connect-type ""' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3' target.new='allow' device.rule='allow id 1d6b:0002 serial "XXXXXX" name "xHCI Host Controller" hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb3" with-interface 09:00:00 with-connect-type ""' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 1d6b:0003 serial "XXXXXX" name "xHCI Host Controller" hash "prM+Jby/bFHCn2lNjQdAMbgc6tse3xVx+hZwjOPHSdQ=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb4" with-interface 09:00:00 with-connect-type ""' device.system_name='/devices/pci0000:00/0000:00:14.0/usb4' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb4' target.new='allow' device.rule='allow id 1d6b:0003 serial "XXXXXX" name "xHCI Host Controller" hash "prM+Jby/bFHCn2lNjQdAMbgc6tse3xVx+hZwjOPHSdQ=" parent-hash "rV9bfLq7c2eA4tYjVjwO4bxhm+y6GgZpl9J60L0fBkY=" via-port "usb4" with-interface 09:00:00 with-connect-type ""' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 046d:c534 serial "XXXXXX" name "USB Receiver" hash "2Tmol95c6dv//0RiOpMlUD2f72+S/vuJuIfLIZ2rNXc=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-3" with-interface { 03:01:01 03:01:02 } with-connect-type "hotplug"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-3' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-3' target.new='allow' device.rule='allow id 046d:c534 serial "XXXXXX" name "USB Receiver" hash "2Tmol95c6dv//0RiOpMlUD2f72+S/vuJuIfLIZ2rNXc=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-3" with-interface { 03:01:01 03:01:02 } with-connect-type "hotplug"' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 30c9:0050 serial "XXXXXX" name "Integrated RGB Camera" hash "F1s4gysKB+YDQiMyLHbVWIxq9IFvWxonx95/bH60F+A=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-5" with-interface { 0e:01:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:01:01 0e:02:01 0e:02:01 fe:01:01 } with-connect-type "not used"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-5' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-5' target.new='allow' device.rule='allow id 30c9:0050 serial "XXXXXX" name "Integrated RGB Camera" hash "F1s4gysKB+YDQiMyLHbVWIxq9IFvWxonx95/bH60F+A=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-5" with-interface { 0e:01:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:02:01 0e:01:01 0e:02:01 0e:02:01 fe:01:01 } with-connect-type "not used"' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 058f:9540 serial "XXXXXX" name "EMV Smartcard Reader" hash "j6z/wqFtA1bZWwBIPmIr/g8KfsEQJ63vpgf4cBcNLbU=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-6" with-interface 0b:00:00 with-connect-type "not used"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-6' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-6' target.new='allow' device.rule='allow id 058f:9540 serial "XXXXXX" name "EMV Smartcard Reader" hash "j6z/wqFtA1bZWwBIPmIr/g8KfsEQJ63vpgf4cBcNLbU=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-6" with-interface 0b:00:00 with-connect-type "not used"' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 06cb:00f9 serial "XXXXXX" name "" hash "N0lDm4gYlYr7p+/D8wvenAbSdB60P/aONrJeyN2djic=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-7" with-interface ff:00:00 with-connect-type "not used"' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-7' type='Device.Present'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.system_name='/devices/pci0000:00/0000:00:14.0/usb3/3-7' target.new='allow' device.rule='allow id 06cb:00f9 serial "XXXXXX" name "" hash "N0lDm4gYlYr7p+/D8wvenAbSdB60P/aONrJeyN2djic=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-7" with-interface ff:00:00 with-connect-type "not used"' target.old='allow' type='Policy.Device.Update'
ago 05 11:08:12 alg-pc usbguard-daemon[1486078]: uid=0 pid=1486073 result='SUCCESS' device.rule='allow id 8087:0026 serial "XXXXXX" name "" hash "Z5csNGxiUukPPZwSHPyUqpVCNagsfOSSNL2CfXhw4IY=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "3-10" with-interface { e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 e0:01:01 } with-connect-type "not used"' device.system_name='/devices/pc [... cut ...]
allow id one-of { 046d:c52b 046d:c534 } name "USB Receiver" with-interface { 03:01:01 03:01:02 } with-connect-type "hotplug" label "XXXXXX"
allow id 045e:07f8 name "Wired Keyboard 600" hash "R+y+7NMVCKLRalharwiNXPe3GG/zIlLg1OXVkIx6fa4=" with-interface { 03:01:01 03:00:00 } label "XXXXXX"
```

Personal information have been partially obfuscated in all shared snippets.

There is a similar report (#543), but my issue happens regardless the computer being docked or not, so I'm opening a new one.

Thank you for your work!

meden, 2023-09-08:
I think this issue is caused by [this commit](https://github.com/USBGuard/usbguard/commit/9b881ed208c9f8ca8515d69aa328dad79708b778). Not sure about its motive, though.

meden, 2023-10-24:
About the motive, I thick the actual motive is [this one](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/735), so I'm not sure anymore whether this bug is valid or not.

commandline-be, 2023-11-07:
Same issue, tried multiple variations of usbguard config. 

After the last config test I just left the screen to auto-lock, on resume all USB devices awoke as expected by policy.
Whenever I manually lock the screen by enabling suspend the "USB 2.1 Hub" is forgotten and must be re-enabled manually.

apt show usbguard 
Package: usbguard
Version: 1.1.1+ds-3

usbguard get-parameter ImplicitPolicyTarget 
block

usbguard get-parameter InsertedDevicePolicy 
apply-policy

meden, 2023-11-08:
@commandline-be, this may actually not be an USBGuard's issue.

I currently solved the bad behavior by disabling Gnome's USB "security":
```
$ gsettings set org.gnome.desktop.privacy usb-protection false
```

Please check [my comment](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/735#note_1877193) on the gnome-setting-daemon bug tracker for further details.

Hope it helps.

commandline-be, 2023-11-08:
With prior experiences I've also tried disabling usb-protection eventually to no avail. 
the gnome-setting conflict with usbguard is quite old by now and seems not easy to resolve

commandline-be, 2023-11-09:
@meden forget what i said, this works only 50% of the time, it seems deeper suspend states (or something, speculating) are working against this (as noted before)

How is your experience by now ?

meden, 2023-11-14:
@commandline-be, as far as I can tell, things are working properly now (with Gnome's USB security disabled). When resuming the laptop, now GDM suggests that I can use the fingerprint to access (although it looks that sometimes it does it only on the second attempt, as I need to hit e.g. ESC a couple of times to "open-close-open" the password box).

Also, I'm not flooded anymore by the notifications coming from all my USB3 HUB's sub-devices.

commandline-be, 2024-04-14:
facing this issue again i can tell it is not working reliably

jimi3, 2025-05-12:
> $ gsettings set org.gnome.desktop.privacy usb-protection false

Run this as sudo too, and manual suspend and wake up worked on mate 24.04