USBGuard/usbguard #596: USBguard blocks allowed device after kernel disconnects and reconnects USB fingerprint reader

ilario, 2023-08-22:
When using [Fprintd](https://gitlab.freedesktop.org/libfprint/fprintd) on Arch Linux on a Lenovo Thinkpad X1-carbon 2nd gen, the internal fingerprint reader "Validity Sensors, Inc. VFS 5011 fingerprint sensor" gets disconnected and reconnected by the kernel when a second swipe is requested while the first one is still running (e.g. when running another instance of `fprintd-verify` while the sensor light is still on from a previous one).

This happens also, sometimes, when unlocking the Gnome screen lock.

For some reason, if USBguard is installed and configured for blocking any USB device that gets connected while the screen is locked by Gnome, sometimes it blocks the fingerprint reader, even if there is an allow rule for it.

**Expected Behavior**

Regardless of the disconnections triggered by the kernel when the fingerprint reader has problems (e.g. when two swipes are requested one close to another manually running fprintd-verify or by Gnome screen lock), USBguard should not block a device in the allowed devices list.

**Current Behavior**

When running a second fprintd-verify while the sensor light is still blinking after a previous fprintd-verify, this message can be seen in dmesg:

```
usb 2-6: USB disconnect, device number 5
usb 2-6: new full-speed USB device number 6 using xhci_hcd
usb 2-6: New USB device found, idVendor=138a, idProduct=0017, bcdDevice= 0.78
usb 2-6: New USB device strings: Mfr=0, Product=0, SerialNumber=1
usb 2-6: SerialNumber: 9585d92de732
usb 2-6: Device is not authorized for usage
```

the same message can be seen when there are problems with the unlocking of the screen lock.

[image]

Then, every time this happens, USBguard locks the fingerprint reader:

```
# usbguard list-devices
[...]
17: block id 138a:0017 serial "9585d92de732" name "" hash "I42u56Qym91l9WYPg8wt7+lqHyWekuV0oMt0Urt15FA=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" via-port "2-6" with-interface ff:00:00 with-connect-type "not used"
```

Even if it has an allow rule (but the rule is not matching it, as would be revealed with the `-d` option):

```
# usbguard list-rules -d
[...]
4: allow id 138a:0017 serial "9585d92de732" name "" hash "I42u56Qym91l9WYPg8wt7+lqHyWekuV0oMt0Urt15FA=" parent-hash "jEP/6WzviqdJ5VSeTUY8PatCNBKeaREvo2OqdplND/o=" with-interface ff:00:00 with-connect-type "not used"
5: allow id 1c4f:0002 serial "" name "USB Keykoard" hash "8xIdLLosrScsXdP9n6C7oIFuMiEA8XjC7H11ZnxTpIA=" parent-hash "ciNFXcCOqVp11Q/H4YmkindpH2vDaD5FkKaCAETeAfg=" via-port "2-2.4" with-interface { 03:01:01 03:00:00 } with-connect-type "unknown"
[...]
```

**Possible Solution**

USBguard should match the allowed device with the correct rule.

**Steps to Reproduce**

1) Block the devices plugged in while the Gnome screen lock is active (following [these instructions](https://wiki.archlinux.org/title/USBGuard#Block_all_USB_devices_by_default)):

`$ gsettings set org.gnome.desktop.privacy usb-protection-level lockscreen`

2) Lock the Gnome screen.
3) Quickly swipe the wrong finger then press exit and exit again.

Or: (untested)

Set the block policy to always
Run twice fprintd-verify without waiting for the blinking fingerprint reader light to stop.

**Context**

Arch Linux

Linux iglaptop 6.4.11-arch2-1 #1 SMP PREEMPT_DYNAMIC Sat, 19 Aug 2023 15:38:34 +0000 x86_64 GNU/Linux

gnome-desktop 1:44.0-1
fprintd 1.94.2-2
usbguard 1.1.2-2

ilario, 2023-08-22:
More info on the original report here: https://bugs.archlinux.org/task/78957

muelli, 2023-08-22:
> For some reason, if USBguard is installed and configured for blocking any USB device that gets connected while the screen is locked by Gnome, sometimes it blocks the fingerprint reader, even if there is an allow rule for it.

GNOME sets the InsertedDevicePolicy to block when the lockscreen is on and manually activates hubs and HIDs.

Could it be that GNOME is blocking the fingerprint reader rather than USBGuard?
We have a report here: https://gitlab.gnome.org/GNOME/gnome-settings-daemon/-/issues/735

Can you try to disable the GNOME functionality, i.e. turn all protection off, and try again?

The problem, IIRC, is that the fingerprint hardware maintains state and needs to be re-set every so often. @benzea can fill in more details.

ilario, 2023-08-31:
Thanks for the answer, yes, deactivating the Gnome feature fixed it.
Closing here and hoping that the Gnome people can fix it soon :)

closed 2023-08-31 by ilario

benzea, 2023-09-01:
Hrm, well, this *is* a problem, seems like I forgot to write the details.

There are two things that can happen at suspend/resume with USB devices:
 1. They get reset and are rediscovered
 2. They keep awake the whole time and are there exactly as before

Now, for 1. the kernel has the PERSIST feature, which hides the fact that the device is rediscovered after resume (and should get an unplug/plug event). This is great for e.g. USB drives, because their state will survive and you do *not* want to loose file system access.

Unfortunately, fingerprint readers have state. We *can* put the reader into scanning mode while suspended which means we will get woken up if it manages to read a print while during suspend. i.e. case #2 above. This does not always work, e.g. the BIOS may turn off USB, in which case we try to do case 2 but actually have case 1.

Userspace needs to know the difference. So, if 2 does not work, we need the unplug/plug event. But, that causes the device to be "new" and USBGuard blocks it.

Not sure, maybe USBGuard needs a kind of "persist" feature of its own to deal with this.

benzea, 2023-09-01:
Then, the problem is that we should always permit fingerprint readers (also solving the resume problem)?

Still need to figure out how to do that. Not sure we want to trust every known reader out there by default … possibly:
 * everything for which the port is marked as internal?
 * anything that has been there from boot (and just reappears after suspend)?

Or, as you say, assume someone generated an `/etc/usbguard/rules.conf`, but not sure we can rely on that …

It feels like there is a problem here. Maybe not suspend/resume related but still something that is problematic in principle.