USBGuard/usbguard #607: Monitoring copied file to a usb key

p1r4t3-s4il0r, 2023-10-26:
Hello, 

I'm using USB guard to prevent users from connecting an untrusted device to their machines (Unix). 

Once the device is trust, I would like to keep logs of which file is copied to a USB key. I would like to have :
 - the username
 - the timestamp
 - the file name
 - an the most important for me : the hash of the file.

I didn't find any docs about this with USB guard, is it doable ? And, if not, what tool can I use to do so ?

Today, I have to go on each machine to authorize a device. I would like to "centralize" the authorizations. I would like to authorize devices regarding the connected user using his LDAP account. Is there any trick for this ? And should I open an other issue for this ?

Thanks.

radosroka, 2023-10-26:
Perhaps (just a wild guess) https://github.com/linux-audit can do something like that.

p1r4t3-s4il0r, 2023-10-26:
It seems a bit overkill
But I'll give it a try and let you know

radosroka, 2023-11-16:
If you don't mind I will close the issue. Feel free to reopen if needed.

closed 2023-11-16 by radosroka