USBGuard/usbguard #608: enabling USBGuard with LDAP

p1r4t3-s4il0r, 2023-10-31:
Hello to everyone. 

I'm using USB guard to prevent users from connecting an untrusted device to their machines (Ubuntu).

But I have to go on each machine to authorize a device. I would like to "centralize" the authorizations. I would like to authorize devices regarding the connected user using his LDAP account. Is there any trick for this ? 

Thanks

p1r4t3-s4il0r, 2023-11-06:
Thanks for your answer. I'm not sure to understand the difference between policy and authorization.

Hmm, yeah I could have a work around collecting the audits to allow a device.

radosroka, 2023-11-06:
You want to probably forward logs with rsyslog or with other logging daemon. You need to manually evaluate the devices but from one centralized place. If needed you can create a rule for specific device and specific host. The collection of the rules is called policy. If stored in LDAP, the daemon will "download" it when it needs to decide what to do with newly connected device. I haven't looked at the code but when you add a rule or change a policy any how the difference can be spotted after new event comes. E.g. insertion of a new device or re-insertion of the already connected device.

p1r4t3-s4il0r, 2023-11-07:
Thanks for your suggestion, I just don't know how to trigger the download of the new policy. 

I don't understand what you mean by "If stored in LDAP the daemon will ...". 
How can the deamon be stored in LDAP ?

Thanks for your time @radosroka.

radosroka, 2023-11-07:
> Thanks for your suggestion.
> 
> I just don't know how to trigger the download of the new policy, I don't understand what you mean by "If stored in LDAP the daemon will ...". How can the deamon be stored in LDAP ?
> 
> Thanks for your time @radosroka.

if policy is stored in LDAP

p1r4t3-s4il0r, 2023-11-07:
Thanks. That's more clear.

radosroka, 2023-11-16:
If you don't mind I'm gonna close this issue. Feel free to reopen if needed.

closed 2023-11-16 by radosroka