USBGuard/usbguard #617: usb-c passthrough USB-HUB are 'forgotten'

commandline-be, 2024-04-14:
Though use of  'gsettings set org.gnome.desktop.privacy usb-protection false' has some effect there is no permanent fix.

Set-up is a Monitor with USB-C and USB-A ports, there is a USB-A (3.1) hub connected to the monitor USB-A port, the usb-c port is connected to the computer. The monitor steering happens over USB-C, the USB-HUB(s) in the monitor are found on the computer.

Whenever the computer reboots are resumes from suspend or hibernate the USB HUBs must be permitted again, the connect-type is reported as 'unknown'

There is one allow-action required for the via-port (N-N.N) and one for the child via-port (N-N.N.n) There is no difference on the with-interface mentioned for these ports. So, though there's some variance in the hashes this setup seems mostly if not entirely stable.

In other words, it seems the connection are forgotten while they're now repeatedly found in the rules.conf file

Is this a known issue ?

bwildenhain, 2024-04-18:
I can confirm forgotten devices connected via a USB-C screen (Dell P3223DE) (USB 3.2 internal Ethernet adapter and USB 2.1 Logitech C930e webcam), while other devices on the same screen work (no additional hub involved).

commandline-be, 2024-11-10:
@secretmango would not report such if it worked would I/we ?