USBGuard/usbguard #618: How to make it autostart?

mYnDstrEAm, 2024-04-21:
On Debian12/KDE I have put usbguard-notifier into Autostarts but the usbguard daemon doesn't start automatically. How to make it autostart? Currently it needs to be started manually with `sudo systemctl start usbguard.service`. I think it would be nice if there was an easy option in a GUI to make it autostart. Mentioned this in #161

mYnDstrEAm, 2024-04-23:
Hidden in the manualpage of usbguard-notifier (`man usbguard-notifier`) there is the info that it can be configured to autostart with `systemctl enable --user usbguard-notifier.service`.

Since there is an Autostarts setting in KDE users would go there to configure autostart if it's not set by default, not prompting the user if it should autostart, and not shown somewhere where the user would see it such as after the first run / the commands that need to be run before it is configured. So as long as Autostarts doesn't show autostarts configured with systemctl (which it should!), I think something should be done on the usbguard side.

jmbit, 2024-05-10:
I don't think this is really an issue for USBguard. usbguard-notifier is a daemon usually run as a systemd service. The correct KDE tool to manage those seems to be systemdgenie. "Autostart" is more for automatically opening your browser on startup and stuff like that.