USBGuard/usbguard #624: USB device (RTL8152) not working after disconnecting and reconnecting

jmbit, 2024-05-10:
Recently my laptop stopped connecting to the Ethernet adapter on my docking station.

After debugging, it seems to be likely connected with usbguard, at least it only appears with usbguard running:

```
[   55.112506] usb 4-2.4: new SuperSpeed USB device number 4 using xhci_hcd
[   55.138040] usb 4-2.4: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=31.11
[   55.138052] usb 4-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[   55.138057] usb 4-2.4: Product: USB 10/100/1000 LAN
[   55.138060] usb 4-2.4: Manufacturer: Realtek
[   55.138062] usb 4-2.4: SerialNumber: 111000001
[   55.138377] r8152-cfgselector 4-2.4: Device is not authorized for usage
[   55.338422] hid-generic 0003:413C:B06E.0008: hiddev0,hidraw2: USB HID v1.11 Device [Dell dock] on usb-0000:04:00.0-2.5/input0
[   55.361942] usb 3-2.3.4: authorized to connect
[   55.367390] hid-generic 0003:413C:B06F.0009: hiddev1,hidraw3: USB HID v1.11 Device [Dell dock] on usb-0000:04:00.0-2.3.5/input0
[   55.367433] usb 3-2.3.5: authorized to connect
[   55.369700] r8152-cfgselector 4-2.4: authorized to connect
[   55.893299] hid-generic 0003:413C:B06F.000A: hiddev1,hidraw3: USB HID v1.11 Device [Dell dock] on usb-0000:04:00.0-2.3.5/input0
```

after stopping usbguard, reloading and triggering udev, the device works:

```
[ 1822.772255] usb 4-2.4: new SuperSpeed USB device number 4 using xhci_hcd
[ 1822.792639] usb 4-2.4: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=31.11
[ 1822.792644] usb 4-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[ 1822.792647] usb 4-2.4: Product: USB 10/100/1000 LAN
[ 1822.792648] usb 4-2.4: Manufacturer: Realtek
[ 1822.792649] usb 4-2.4: SerialNumber: 111000001
[ 1822.792691] hid-generic 0003:413C:B06E.000C: hiddev0,hidraw2: USB HID v1.11 Device [Dell dock] on usb-0000:04:00.0-2.5/input0
[ 1823.172429] r8152-cfgselector 4-2.4: reset SuperSpeed USB device number 4 using xhci_hcd
[ 1823.195519] r8152 4-2.4:1.0: firmware: direct-loading firmware rtl_nic/rtl8153b-2.fw
[ 1823.208563] r8152 4-2.4:1.0: load rtl8153b-2 v2 04/27/23 successfully
[ 1823.240690] r8152 4-2.4:1.0 eth0: v1.12.13
[ 1823.260815] r8152 4-2.4:1.0 enxc03eba2c45fd: renamed from eth0
[ 1823.275938] usb 3-2.3.5: new high-speed USB device number 6 using xhci_hcd
[ 1823.393632] usb 3-2.3.5: New USB device found, idVendor=413c, idProduct=b06f, bcdDevice= 1.01
[ 1823.393636] usb 3-2.3.5: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 1823.393637] usb 3-2.3.5: Product: Dell dock
[ 1823.401661] hid-generic 0003:413C:B06F.000D: hiddev1,hidraw3: USB HID v1.11 Device [Dell dock] on usb-0000:04:00.0-2.3.5/input0
[ 1823.501443] userif-3: sent link down event.
[ 1823.501459] userif-3: sent link up event.
[ 1824.508334] hid-generic 0003:413C:B06F.000E: hiddev1,hidraw3: USB HID v1.11 Device [Dell dock] on usb-0000:04:00.0-2.3.5/input0
[ 1826.168673] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[ 1826.657271] r8152 4-2.4:1.0 enxc03eba2c45fd: carrier on
[ 1826.857790] userif-3: sent link down event.
[ 1826.857818] userif-3: sent link up event.
[ 1830.642530] Lockdown: systemd-logind: hibernation is restricted; see man kernel_lockdown.7
[ 1836.044761] userif-3: sent link down event.
[ 1836.044789] userif-3: sent link up event.
```

Device in rules.conf:

```
allow id 0bda:8153 serial "111000001" name "USB 10/100/1000 LAN" hash "iqKSyA0daD0B2vQN2es7JZWYGzQscGtQaWDXT7AE+5s=" parent-hash "vIleUcpJPcyXx2L/6fpGeWmvkBaZV6qqpMKh4Hv4gKs=" with-interface { ff:ff:00 02:06:00 0a:00:00 0a:00:00 } with-connect-type "unknown"
```

lafrenchpatate, 2024-06-11:
Hello,

I'm having the exact same issues as you. When usbguard is working, everything on my dock seems to be working perfectly fine EXCEPT the RJ45 port.

I have the exact same issue if i'm using an USB-C RJ45 adapter.

If i disable usbguard and reboot, both devices allow me to connect to my network through the RJ45.

Did anyone found a workaround?

lafrenchpatate, 2024-06-11:
After further testing it seems that the RJ 45  is functionnal only if the device is plugged on the machine when it boots.

For example:

- my machine is turned on and usbguard is active
- I plug in my adapter with a RJ 45 connected to it
- USBGuard sees the device and allows it
- It doesn't connect me to the network
- I reboot the machine
- After rebooting the RJ45 adapter is working and i'm connected to the network
- If i unplug the USB adapter from the machine and replug it, it doesn't work anymore

I tried this with a DELL dock, a DELL USB-C adapter and a UNITEK adapter, all behaving as described.

I