USBGuard/usbguard #634: USBGuard disabled all peripherals

123J1232, 2024-07-19:
I recently acquired a 2011 iMac with Linux Mint installed, the apple mouse was not recognised so i plugged in a working usb mouse which also was not recognised.  I looked up potential fixes and installed USBGuard which promptly locked all USB devices out and I am now unable to use a USB mouse or keyboard.  I have downloaded Linux Mint Xfce Edition and made a bootable usb stick with Etcher but when inserted into the iMac it is not recognised when i reboot it and with no keyboard/mouse input i cannot get to the boot menu.  The iMac just ignores all input and boots to the previous Linux Mint log in screen that was installed when I received it.  I do not know Linux so please give detailed instructions on how to proceed.  TiA

muelli, 2024-11-29:
Sounds like a duplicate of https://github.com/USBGuard/usbguard/issues/268 or rather https://github.com/USBGuard/usbguard/pull/267

123J1232, 2024-11-29:
> Sounds like a duplicate of https://github.com/USBGuard/usbguard/issues/268 or rather https://github.com/USBGuard/usbguard/pull/267

I read those replies is there a way out of this as those replies mention that there is an updated version but they do not advise on how to resolve if usb devices are not working unless I'm missing something?