USBGuard/usbguard #638: Unable to connect to MTP mobile device.

fxzxmicah, 2024-10-13:
Modern mobile phones are using MTP as the transmission protocol, and for security reasons, it is necessary to manually switch to this mode on the phone. However, after using USBGuard, the MTP device did not appear on the computer.

fxzxmicah, 2024-10-13:
[usbguard.log](https://github.com/user-attachments/files/17355872/usbguard.log)

fxzxmicah, 2024-10-21:
I can't even use adb to connect my phone.

secretmango, 2024-11-07:
Same here

Fedora 40, `usbguard usbguard-selinux usbguard-notifier` installed

Phone is a Google Pixek 6a with GrapheneOS

When connecting, it always shows up as a new device which I need to allow. Same for every usb mode, which also always appears as a new device.

I assume that this is because of some form of randomization which would be needed turned off.

https://discuss.grapheneos.org/d/17175

secretmango, 2024-11-08:
I have found the solution

The usbguard-notify prompt seems to be a temporary allowance. Only the initial setup allows the devices permanently.

Connect your phone with the mode you want.

```
usbguard list-devices
# for example
usbguard allow-device -p 12
```

repeat this for every USB mode, i.e. filesharing, webcam, usb tethering

MTP works for me