USBGuard/usbguard #644: Keyboard and mouse disabled after reboot

nturns, 2024-11-29:
I had this issue on Kali and now Debian 12. The keyboard and mouse are blocked after reboot. Have to boot into safe mode and systemctl disable usbguard and usbguard-dbus. Then regular boot enable again. It doesn't happen on every reboot. If I enable usbguard after I log in it is OK.  I think it has a timing issue if usbguard is enabled too soon in the booting process. Is there a way to change when the system enables usbguard?

muelli, 2024-11-29:
Sounds like a duplicate of https://github.com/USBGuard/usbguard/issues/268 or rather https://github.com/USBGuard/usbguard/pull/267

rsramkis, 2024-12-14:
Out of curiosity, what directory is your rules.conf file in? Also are the permissions set to 600?

nturns, 2024-12-14:
Yes.

/etc/usbguard# ls -l
total 20
drwxr-xr-x 2 root root 4096 Nov 28 00:42 IPCAccessControl.d
-rw------- 1 root root  922 Nov 28 00:42 rules.conf
drwx------ 2 root root 4096 Nov 22  2022 rules.d
-rw------- 1 root root 6652 Nov 22  2022 usbguard-daemon.conf

rsramkis, 2024-12-14:
Can you check the usbguard-daemon.conf file and see if you have these settings:
```
# RuleFile=/path/to/rules.conf
#
RuleFile=/etc/usbguard/rules.d/rules.conf

# RuleFolder=/path/to/rulesfolder/
#
RuleFolder=/etc/usbguard/rules.d/
```
If yes, then you should move your rules.conf file into the rules.d directory.

Think you ran into what I identified here "https://github.com/USBGuard/usbguard/issues/645".

nturns, 2024-12-14:
My usbguard-daemon.conf has RuleFile pointing to where it actually is.

```
RuleFile=/etc/usbguard/rules.conf

RuleFolder=/etc/usbguard/rules.d/
```
The rules.d directory is empty.

I moved rules.conf into rules.d/ and I changed "RuleFile=" to point to it, thus setting it up like yours. When I "systemctl start usbguard" I lost both mouse and keyboard. That was unexpected. As I mentioned I originally was able to start usbguard from the command line and it worked OK.

Checking other items, file rules.conf lists two host controllers and the mouse and keyboard.

In usbguard-daemon.conf the following is correct.

`IPCAccessControlFiles=/etc/usbguard/IPCAccessControl.d/`

rsramkis, 2024-12-15:
Thought I would let you know  that when I booted my systems today I also lost my external USB keyboard and mouse plugged into my dock. So I have updated my defect ticket for the USBGuard team to investigate.

A temporary workaround for me was to downgrade my Linux Kernel from  6.12 to 6.11.9. So I am not sure if the 6.12 Kernel Branch introduced some new  feature which is causing an issue.