USBGuard/usbguard #663: Any chance for a new release USBGuard 1.1.4 off of `master`?

hartwork, 2025-04-28:
Hi!

Is there a chance for a new release off `master` e.g. for the Protobuf v30 fixes and support for Catch2 v3?
A new release would help distros to not need to cherry-pick the post-1.1.3 fixes.

Thanks and best, Sebastian

CC @Cropi

hartwork, 2025-05-11:
@Cropi please please 🙏

Cropi, 2025-05-19:
Hello, I will try to spare some time to do the official build soon (I also need it :D). Allow me some time to process it, I am a bit sick right now.

semenzato, 2025-05-20:
Thank you @hartwork, I hadn't seen this.  Get well soon @Cropi!

Cropi, 2025-07-15:
Sorry for the delay. The new release is now available, https://github.com/USBGuard/usbguard/releases/tag/usbguard-1.1.4

hartwork, 2025-07-15:
@Cropi thank you! 👍 👍